luc-github/esp3d.io #16: Using ESP3D in JGMaker Magic 3D printer

urobot2011, 2023-05-25:
Please use this issue only for esp3d.io web site issue 
for [ESP3D](https://github.com/luc-github/ESP3D) /  [ESP3DLib](https://github.com/luc-github/ESP3DLib) /  [ESP3D-TFT](https://github.com/luc-github/ESP3D-TFT)  /  [ESP3D-WEBUI](https://github.com/luc-github/ESP3D-WEBUI)

Please describe what whould be an enhancement

-------------------------
It would be nice to have a manual on 'Using ESP3D in JGMaker Magic 3D printer' on the wiki. Reference photo:

https://images.chesscomfiles.com/uploads/v1/images_users/tiny_mce/urobot2011chess/phpSXB5Lz.jpg

luc-github, 2023-05-25:
pr are welcome

luc-github, 2023-05-25:
if you do not know how to do a pr, you  provide textand image in issue I wil add it to web site

urobot2011, 2023-05-25:
photo: https://images.chesscomfiles.com/uploads/v1/images_users/tiny_mce/urobot2011chess/phpSXB5Lz.jpg. Solder it here and level down (5v to 3.3v) to the esp32/8266 and connect it.

luc-github, 2023-05-28:
I have updated the section : https://esp3d.io/esp3d/v3.x/installation/wiring/index.html#jgmaker-magic-3d-printer

Thank you

closed 2023-05-28 by luc-github